openxrlab/xrslam #2: Windows?

antithing, 2022-09-01:
Hi! This code looks great, thank you for making it available. 

Would it be possible to compile and run on Windows? 
Does it run online, for example with a realsense d455?

Thanks!

panxkun, 2022-09-03:
it seems that the SLAM pc-player can be built successfully on windows and start up , but the data stream was blocked. Take a try this command

D:\testing\xrprimer-main\xrslam-main\build\xrslam-pc\player\Release>xrslam-pc-player -c configs/euroc.yaml --tum trajectory.tum euroc:///D:/datasets/MH_05_difficult/**mav0**

dataset path need set to the `mav0` folder

antithing, 2022-09-04:
Hi, thank you for getting back to me. I used the command as you suggest, but i see the same result. The button has no effect, and the slam does not run. Thank you again

panxkun, 2022-09-05:
Sorry for the unfamiliar with the command line mode on Windows. It is better to use absolute paths for reading files, including the configuration files `euroc.yaml`

antithing, 2022-09-05:
I have adjusted the command to be:
`
xrslam-pc-player -c D:/testing/xrprimer-main/xrslam-main/build/xrslam-pc/player/Release/configs/euroc.yaml --tum trajectory.tum euroc:///D:/datasets/MH_05_difficult/mav0`

But I see the same issue! I might try hardcoding the filepaths in the c++

antithing, 2022-09-05:
Hi again,

Stepping through the code, it appears that this function:

`    void load(const std::string &filename) {
        items.clear();
        if (FILE *csv = fopen(filename.c_str(), "r")) {
            char header_line[2048];
            fscanf(csv, "%2047[^\r]\r\n", header_line);
            char filename_buffer[2048];
            CameraData item;
            while (!feof(csv)) {
                memset(filename_buffer, 0, 2048);
                if (fscanf(csv, "%lf,%2047[^\r]\r\n", &item.t,
                           filename_buffer) != 2) {
                    break;
                }
                item.t *= 1e-9;
                item.filename = std::string(filename_buffer);
                items.emplace_back(std::move(item));
            }
            fclose(csv);
        }
    }

`

Does not work on Windows. Even with a hardcoded file path, it still returns empty.

Thanks!

huanggan52, 2022-09-07:
Perhaps it is because the symbol at the end of each line of text is expressed differently in linux and windows, you can replace "[^\r]" with "[^\n]" to have a try，and three symbols need to be modified in the above file .

antithing, 2022-09-07:
Thank you! That has it running. however, i now get a crash here:

`xrslam->track_accelerometer(t, a.x(), a.y(), a.z());`

`Unhandled exception at 0x00007FFEDC12474C in xrslam-pc-player.exe: Microsoft C++ exception: cv::Exception at memory location 0x00000055E5B5DC80.`

itsuhane, 2022-09-08:
`cv::Exception` suggests something wrong with OpenCV. OpenCV is only used for image loading/keypoint tracking. Can you confirm that the images are correctly loaded?

antithing, 2022-09-08:
Hi, I have now hardcoded the paths, and can confirm that the image paths are correct (i am showing the images using imshow) My reader looks like this:

[image]

So it appears the data is loaded correctly, however I still see the crash. 

If i comment out this line:

`xrslam->track_accelerometer(t, a.x(), a.y(), a.z());`

the application runs, and i see the images update, but the 3d view stays empty.

Thanks!

itsuhane, 2022-09-09:
The VIO will not initialize itself if there are no IMU measurements, hence outputs zero poses.

Did you still get `cv::Exception` for the crash? If so, I recommend you check opencv related code in [`xrslam-extra/src/xrslam/extra/opencv_image.cpp`](https://github.com/openxrlab/xrslam/blob/6590f3fe3a923c3339c2d33ed50ace7cf9409bd3/xrslam-extra/src/xrslam/extra/opencv_image.cpp).

antithing, 2022-09-09:
Hi, Ah, indeed, the crash is here:

`clahe()->apply(image, image);`

If i comment that line, it crashes on the next :

`buildOpticalFlowPyramid(image, image_pyramid, Size(21, 21),
                            (int)level_num(), true);`

It seems like the image is not loading, but I can view it, and print the type and channels. Very strange! Can you think of anything else i can look at here?

itsuhane, 2022-09-10:
Back to the dataset reader for euroc, it reads a list file, extracting timestamp and file name from the line.

Maybe you can take a close look at the extracted file name string, see if especially any invisible characters (CRLF) in the path string.

Best,
Jinyu

antithing, 2022-09-12:
Hi again, if i edit the following to show the image:

`void OpenCvImage::preprocess() {

    cv::imshow("test", image);
    cv::waitKey(1);

    clahe()->apply(image, image);
    image_pyramid.clear();

    buildOpticalFlowPyramid(image, image_pyramid, Size(21, 21),
                            (int)level_num(), true);
}`

It displays the image correctly, but crashes on the ` clahe()->apply(image, image);` line. Strange!

So the images are being loaded, but the function crashes

itsuhane, 2022-09-13:
I have no experience with windows systems so I need your help for some baseline test.

Can you modify the `main` function in file `xrslam-pc/player/src/main.cpp` to something like this?

```cpp
// starting at line 251:
int main(int argc, char *argv[]) {
    // add the following:
    using namespace cv;
    Mat image = imread(/* an image path known to load */, IMREAD_GRAYSCALE);
    Ptr<CLAHE> clahe = createCLAHE(6.0, cv::Size(8, 8));
    clahe->apply(image, image);
    // ...
```

What it does is simply reading an image and running the CLAHE algorithm (before any real XRSLAM thing kicks in).
If that makes xrslam-pc crash, then there should be something wrong with your local OpenCV build.

Meanwhile, I'll ask the team to see if they can provide a precompiled windows binary for XRPrimer, a tested OpenCV should come with it. So far I haven't managed to build XRPrimer on windows, which is blocking me from further investigating this issue.

Best,
Jinyu

antithing, 2022-09-16:
Hi, ah apologies! I indeed had an opencv issue! I was linking a opencv 460 lib instead of the included _ext version. 

It is now running as expected on euroc.

next step, to link a live D455  as input! 

Quick question, where can i see the frame time? Are there any fps counters in the code? I am looking for 50 fps, is this an achievable goal?

Thanks again

itsuhane, 2022-09-19:
That's great.

The D455 sensor comes with RGB-D and IMU data. Currently, XRSLAM only handles (monocular) RGB+IMU, and should support D455 given a correct calibration. You can have a look at the calibration file (`configs/euroc.yaml` for example) to know the required calibration data. In the future, we may support tracking using RGB-D data in XRSLAM.

The time spike is indeed from processing the keyframes (BA of the sliding window). Currently we don't have a frame-rate goal, and the solving process is not optimized. For example, we fully build the whole optimization problem each time, which is definitely not optimal. We will improve the design in the future, meanwhile users can customize the code to fit their need. In your case, you need to optimize the solving, and maybe change the strategy used in the sliding window.

Since the runtime error is resolved, I'll close this issue.

closed 2022-09-19 by itsuhane