openxrlab/xrslam #14: The data recorder for iphone and android cameras and IMU external parameter calibration

wangni-bupt, 2023-01-04:
Could you please provide the data recorder for iphone Recently, I want to test on some phones of android and iphone, and the external parameter calibration adopts the Kalibr method suggested by you.
There are several open source projects that find loggers for android phones:
https://github.com/DavidGillsjo/VideoIMUCapture-Android
And the iphone's recorder:
https://github.com/Varvrar/ios_logger

At present, there is no test for the above two methods, so I would like to ask if you have a special data collector and can you provide it?
Thank you very muchand android cameras and IMU external parameter calibration

ZibYang, 2023-01-06:
You can find our data collector at http://doc.openxrlab.org.cn/openxrlab_document/ARDemo/ARdemo.html, and we only provided a recorder that can run on iOS devices for now.

wangni-bupt, 2023-01-06:
Thank you for your answer

closed 2023-01-06 by wangni-bupt