openxrlab/xrslam #18: 有关ios下的imu参数问题

lcc157, 2023-02-16:
您好，这是一个非常棒的项目，有关imu有个问题请教一下sigma_u: 0.5，sigma_w: 4.e-8，sigma_a: 4.e-6，sigma_bg: 1.0e-10，sigma_ba: 1.0e-8，他们的IMU是多少HZ下标定得到的呢，是100HZ吗

wangnancpp, 2023-03-02:
是的，我们是在100Hz下标定的，iOS获取到的IMU最大帧率也是100Hz

lcc157, 2023-03-02:
谢谢，非常感谢您的回答，我还有两个疑问，
（1）在ios下/xrslam-ios/visualizer/src/Motion.swift 获取原生的加速度和角速度，加速度默认的单位是g-unit, 转换到m/s^2，不应该是乘9.80665嘛，在您的工程用的GRAVITY_NOMINAL = -9.80665
（2）在[xrslam]/xrslam/src/xrslam/core/detail.cpp  :track_gyroscope 和track_accelerometer 我理解为根据时间戳对齐acc和gyro, 如果我想改用 motion这个api获取到的加速度和角速度，只有一个时间戳，是否可以忽略track_gyroscope 和track_accelerometer，motion这个api获取到的加速度和角速度来自于ios_logger工程（https://github.com/Varvrar/ios_logger/blob/master/ios_logger/ViewController.mm）

`-(void)outputDeviceMotionData:(CMDeviceMotion *)devmotdata
{
    if (isRecording && devmotdata != nil)
    {
        double msDate = bootTime + devmotdata.timestamp;
        
        CMQuaternion quat = devmotdata.attitude.quaternion;
        [logStringMotion appendString: [NSString stringWithFormat:@"%f,%f,%f,%f,%f\r\n",
                                       msDate,
                                        quat.w,
                                        quat.x,
                                        quat.y,
                                        quat.z]];
        
        CMRotationRate rotr = devmotdata.rotationRate;
        CMAcceleration grav = devmotdata.gravity;
        CMAcceleration usracc = devmotdata.userAcceleration;
        [logStringMotARH appendString: [NSString stringWithFormat:@"%f,%f,%f,%f,%f,%f,%f,%f,%f,%f,%f\r\n",
                                       msDate,
                                        rotr.x,
                                        rotr.y,
                                        rotr.z,
                                        grav.x,
                                        grav.y,
                                        grav.z,
                                        usracc.x,
                                        usracc.y,
                                        usracc.z,
                                        devmotdata.heading]];
        
        CMCalibratedMagneticField calmagnfield = devmotdata.magneticField;
        [logStringMotMagnFull appendString: [NSString stringWithFormat:@"%f,%f,%f,%f,%d\r\n",
                                       msDate,
                                        calmagnfield.field.x,
                                        calmagnfield.field.y,
                                        calmagnfield.field.z,
                                        calmagnfield.accuracy]];
    }
}`

huanggan52, 2023-03-06:
1）ios读取原生的加速度和角速度时，竖屏时坐标系为右上里，静止时y值约为-1，而xrslam处理的数据向下为正，所以需要乘以 -g, 您也可以参照euroc的imu数据，开始阶段朝重力的方向的值均为正数
2）只有一个时间戳并没有关系，实际上我们读取euroc的imu数据也是同一个时间戳，即 t0, acc0, gyr0，但为了适应不同时间戳的情况，我们才将其分开，所以并不影响。可以 track_gyroscope(to, gyr0), track_accelerometer(t0, acc0)

closed 2023-03-16 by huanggan52