openxrlab/xrslam #26: More then a single camera

pablovela5620, 2023-03-04:
Considering most VR headsets use 2+ headsets (4 on quest 2/ quest pro/ pico 4/ etc...) are there any plans to be able to add more than just monocular visual-inertial slam?

wangnancpp, 2023-11-08:
VR devices need to design a customized VIO algorithms for sensors. Currently, we have no plans to develop and release the VIO algorithm for head-mounted devices.

pablovela5620, 2023-11-09:
> VR devices need to design a customized VIO algorithms for sensors. Currently, we have no plans to develop and release the VIO algorithm for head-mounted devices.

Makes sense, what about for stereo inertial systems like intel realsense T265 or D435i? Similar to how DroidSLAM allows for monocular/stereo/depth systems (without imu unlike this implementation). Is the target only monocular vio systems? Thanks for the response