openxrlab/xrslam #42: 请教个问题，我不能成功的编译，环境是正常的。

uer7e67, 2024-01-08:
我使用了unbuntu，
执行  cmake -S. -Bbuild -DBUILD_EXTERNAL=ON -DCMAKE_BUILD_TYPE=Release -DENABLE_PRECOMPILED_HEADERS=OFF && cmake --build build --target install -j8  这个命令的时候，
`
gmake[2]: *** [CMakeFiles/pybind11-populate.dir/build.make:99: pybind11-populate-prefix/src/pybind11-populate-stamp/pybind11-populate-download] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/pybind11-populate.dir/all] Error 2
gmake: *** [Makefile:91: all] Error 2

CMake Error at /usr/share/cmake-3.22/Modules/FetchContent.cmake:1087 (message):
  Build step for pybind11 failed: 2
Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/FetchContent.cmake:1216:EVAL:2 (__FetchContent_directPopulate)
  /usr/share/cmake-3.22/Modules/FetchContent.cmake:1216 (cmake_language)
  /usr/share/cmake-3.22/Modules/FetchContent.cmake:1259 (FetchContent_Populate)
  cmake/external/pybind11.cmake:17 (FetchContent_MakeAvailable)
  CMakeLists.txt:71 (include)`

就卡到这里了。 这是为什么，c++，gcc，cmake，都符合要求

huanggan52, 2024-01-25:
cmake -S. -Bbuild -DBUILD_EXTERNAL=ON -DCMAKE_BUILD_TYPE=Release -DENABLE_PRECOMPILED_HEADERS=OFF && cmake --build build --target install -j8 这个命令是编译xrprimer，请先确保已经切换分支到xrslam-opencv3.4.7

uer7e67, 2024-01-29:
感谢回复。我已经解决了这个问题是因为有个东西没有拉取下来的原因，还有个问题求助，那个数据采集器是哪个工程？我并没有在OpenXRLab中找到