openxrlab/xrslam #54: Contribution FIxed installation

OctaAIVision, 2024-10-14:
I hope this message finds you well. I have encountered some installation bugs and errors while working with the XRSLAM repository. I would like to contribute by providing the necessary steps and solutions to resolve these issues.

Would you be open to supporting this contribution and helping me ensure that the fixes align with the repository's standards? I believe this could improve the installation process for others facing similar challenges.

Looking forward to your response.

Best regards,
Kalami

panxkun, 2024-11-03:
any more detail about the bugs or errors?

OctaAIVision, 2024-11-04:
In installation steps there is some bugs. for example in opencv build or other third parties.
if u are willing I can submit a merge request.