openxrlab/xrslam #56: Linux Run error

ebigMan, 2024-11-13:
There were no problems during the installation and compilation process. I also switched to the xrslam-opencv3.4.7 branch. The file structure is as follows. 
.
├── data
├── xrprimer
└── xrslam

When executing the ./build/xrslam-pc/player/xrslam-pc-player -sc configs/euroc_slam.yaml -dc configs/euroc_sensor.yaml --tum trajectory.tum euroc://../data/EuRoC/MH_01_easy/mav0 command, 

the prompt is:

terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(3.4.7) /home/user/xrprimer/_ext/opencv/modules/imgproc/src/undistort.dispatch.cpp:177: error: (-215:Assertion failed) dst.data != src.data in function 'undistort'

How to solve it?

Idea-in-Dream, 2024-11-14:
Version rollback
commit 74d7fa6cfc21897d5a7d28507a5801a0533eefcc (tag: v0.6.0)
this available

ebigMan, 2024-11-14:
Thanks @Idea-in-Dream, it can run,After executing the command, there will be no visual interface, right? I only see relevant data output in the terminal and local .tum format files. Does the visual interface of the demo need to be implemented separately?

ebigMan, 2024-11-14:
Thanks @Idea-in-Dream, it can run, if possible, I would also like to ask how the real-time performance of this algorithm is. When I run it through ROS, I find that there is a large delay in trajectory generation. Thank you very much for your help!

Idea-in-Dream, 2024-11-14:
The cpu usage is very high. Do you have this problem?

ebigMan, 2024-11-15:
I'm sorry, no, my CPU utilization is only a few percent. My PC has many cores, but the utilization is not high, so the bag has finished playing but the trajectory has not been drawn. Is there any relevant setting in the multi-threaded code you mentioned? I'm just starting to use this code and I'm not familiar with it yet. Thank you!

panxkun, 2024-12-15:
If you find that running EuRoC with the default configuration results in high CPU usage or low frame rates, you can adjust parameters in the config, such as the number of feature points and their distance. You’ll notice everything runs normally after the adjustment. Under normal circumstances, it should have low CPU usage and real-time frame rates (faster than VINS).