openxrlab/xrslam #57: What config was used for ADVIO?

ArmandB, 2024-11-25:
Hello, I'm trying to reproduce the results of RD-VIO on the ADVIO dataset but am having trouble finding the configuration file you all used for it.

If I'm not missing something, would you all be open to sharing it?

Thank you for your help in advance!:D

panxkun, 2024-12-15:
Sorry for the delay in response. We do have a configuration file and a processing script for the raw ADVIO data. We realized we missed submitting these files, but we’ll address this immediately and get back to you with an update.

huanggan52, 2024-12-16:
@ArmandB Could you please provide your email address? I would like to send you the script and configuration file needed to run on the ADVIO dataset.

ArmandB, 2024-12-16:
No worries. Thank you guys much! My email is armandb@utexas.edu

closed 2024-12-17 by huanggan52

rafaelspring, 2025-01-03:
@huanggan52 @panxkun 
I would also like to run RD-VIO on the ADVIO dataset. Any chance you can make the script and config file public?

weating, 2025-02-17:
@panxkun @huanggan52 May I ask if you could also send me a copy of the data processing script and configuration files used for running ADVIO? Thank you very much. Here is my email: jerryzhsmart@gmail.com.

panxkun, 2025-02-17:
Hi @rafaelspring @weating , this is the config and script to start with ADVIO datasets https://drive.google.com/file/d/1KZ925Oncxv56CriEf1MszXGU54J5fNwv/view?usp=sharing

weating, 2025-02-17:
@panxkun 非常感谢！

weating, 2025-02-17:
@panxkun 不好意思，我使用您的配置文件，如下指令./build/xrslam-pc/player/xrslam-pc-player -sc configs/euroc_slam.yaml -dc configs/advio-01-12.yaml --tum trajectory_ADVIO02.tum euroc:///home/public/add_disk/peibozun/experimental_data/advio_eurocType/advio-02，系统提示terminate called after throwing an instance of 'xrslam_0_5_0::extra::YamlConfig::ConfigMissingException'
  what():  config "cam0.resolution" is mandatory   resolution是什么参数？

weating, 2025-02-17:
@panxkun 不好意思，我使用您的配置文件，如下指令./build/xrslam-pc/player/xrslam-pc-player -sc configs/euroc_slam.yaml -dc configs/advio-01-12.yaml --tum trajectory_ADVIO02.tum euroc:///home/public/add_disk/peibozun/experimental_data/advio_eurocType/advio-02，系统提示在抛出 'xrslam_0_5_0：：extra：：YamlConfig：：ConfigMissingException'
的实例后调用了终止 what（）： config “cam0.resolution” is mandatory resolution是什么参数？

weating, 2025-02-18:
@panxkun 不好意思，我在自行调整了您的配置文件格式后运行程序时结果出现了较大的误差，请问您是否可以再给我一份现在可用的正确的配置文件来运行ADVIO？非常感谢您

panxkun, 2025-02-18:
> [@panxkun](https://github.com/panxkun) 不好意思，我在自行调整了您的配置文件格式后运行程序时结果出现了较大的误差，请问您是否可以再给我一份现在可用的正确的配置文件来运行ADVIO？非常感谢您

我们后续并没有维护ADVIO数据集的相关部分，因此现在没有可以直接用的配置文件。但是你可以将代码切换到最开始的commit版本，这样就可以直接使用这份配置文件了

weating, 2025-02-19:
@panxkun 非常抱歉继续打扰您，但是我的问题还是没有解决QAQ。我把代码切换到0.4.0版本然后重新编译后确实可以直接使用您的配置文件，而且我用您的脚本转化了ADVIO数据集。我在无GUI环境使用您的程序运行了advio-02和advio-12，但是我仍然无法得到正确的输出文件，输出的轨迹和gt中的轨迹完全不一致。

weating, 2025-02-19:
@panxkun 非常抱歉继续打扰您，但是我的问题还是没有解决QAQ。我把代码切换到0.4.0版本然后重新编译后确实可以直接使用您的配置文件运行我用您的脚本转化的ADVIO数据集，根据0.4.0版本文档的要求，我还修改xrslam-pc/player/src/euroc_dataset_reader.cpp L55,56 的参数和您的配置文件一致。我在无GUI环境使用您的程序运行了advio-02和advio-12，但是我仍然无法得到正确的输出文件，输出的轨迹和gt中的轨迹完全不一致。我一定是哪里搞错了，但是我折腾了一天也没研究出来为什么。

panxkun, 2025-02-20:
> [@panxkun](https://github.com/panxkun) 非常抱歉继续打扰您，但是我的问题还是没有解决QAQ。我把代码切换到0.4.0版本然后重新编译后确实可以直接使用您的配置文件运行我用您的脚本转化的ADVIO数据集，根据0.4.0版本文档的要求，我还修改xrslam-pc/player/src/euroc_dataset_reader.cpp L55,56 [...]

你好，这个现象可能来自于旧版本代码（v0.4.0）逻辑层面缺少对advio的支持，因为旧版本代码仓库和论文版本代码有区别。但是在后续的更新中又添加上了对应的支持。你前面提到了最新代码无法得到正确的结果，这个我们也暂时无法确认。你可以将新版和旧版的dataloader进行对比更新一下，类似这种地方的更新
https://github.com/openxrlab/xrslam/blob/d1ec7d3401ea31adfae40d429b7a5a88a094f5d3/xrslam-pc/player/src/IO/euroc_dataset_reader.cpp#L31-L38
这是对advio数据集的一个必要的改动

weating, 2025-02-21:
@panxkun 非常感谢您耐心的教导，现在我已经得到了接近正确的结果。最后我想知道从哪里可以找到您们的论文来学习，再次感谢您的耐心教导

panxkun, 2025-02-22:
你好，这是相关文章信息

@article{li2023rd,
  title={RD-VIO: Robust Visual-Inertial Odometry for Mobile Augmented Reality in Dynamic Environments},
  author={Li, Jinyu and Pan, Xiaokun and Huang, Gan and Zhang, Ziyang and Wang, Nan and Bao, Hujun and Zhang, Guofeng},
  journal={arXiv preprint arXiv:2310.15072},
  year={2023}
}

欢迎参考引用